joeycastillo/Sensor-Watch #2: Question on repeated segment IDs

jamesbowman, 2021-08-07:
Looking at:

https://github.com/joeycastillo/Sensor-Watch/blob/8a49b9ae2be26cf6e4a3ab4283216eaf61117986/watch-library/watch/watch.c#L124

This line gives the segment addresses for character position 2 on the display.
Starting at segment 0, it specifies:

```
49: SLCD_SEGID(1, 9)
09: SLCD_SEGID(0, 9)
89: SLCD_SEGID(2, 9)
49: SLCD_SEGID(1, 9)
0a: SLCD_SEGID(0, 10)
c0: None
49: SLCD_SEGID(1, 9)
```

It makes sense that segment 5 is missing -- it's not present on that digit -- but why is (1,9) repeated three times? Shouldn't all the IDs be unique?

Or are they wired together in the hardware? Since this digit only ever shows blank, 0, 1, and 2, it makes sense that the original makers would have wired all the horizontal segments together.

Just confirming my understanding, thanks!

joeycastillo, 2021-08-07:
Your understanding is right, they are wired together in the hardware! There are a few other digits that do this: positions 4 and 6 have their top and bottom segments wired together (I have a “lowercase” 7 in the character set to work around this), and position 1 has two segments like this (left and right sides can only be addressed together). Yesterday I was actually working on a segment map; I’ll update this issue when I add it to the repository.

jamesbowman, 2021-08-07:
Got it, thanks. Here's the segment map I've got so far:

[image]

josecastillo, 2021-08-07:
Oh nice! I can clarify the other segments, just added methods for lighting up indicators: https://github.com/joeycastillo/Sensor-Watch/blob/f102be48642235a9d72b2159cc1c32cc83d0de8a/watch-library/watch/watch.c#L137

And the colon: 

https://github.com/joeycastillo/Sensor-Watch/blob/f102be48642235a9d72b2159cc1c32cc83d0de8a/watch-library/watch/watch.c#L187

josecastillo, 2021-08-07:
Positions 0 and 1 have an the eighth segment in the mapping. For position 0, (1,14) is the vertical bar. For position 1, (0,12) is the extra top left segment. 

I forget what that little bottom left segment is on position 0 (I have it on the diagram at the shop) but none of the characters use it since it would require a ninth bit, and it's not really that useful.

jamesbowman, 2021-08-07:
Aha, thanks -- all making sense. And the renders from the demo app look correct:

[image]

[image]

and 0123456789 is legal too:

[image]

joeycastillo, 2021-08-07:
whoa awesome!!

joeycastillo, 2021-08-07:
ok! I added a segment map to the documentation [here](https://joeycastillo.github.io/Sensor-Watch-Documentation/segmap.html) (hover over segments to see the mapping), and added some more notes to the segment map and character set in the watch.c file.

Would be stoked to check out your emulation! I was documenting what each of the characters in the character set look like in the various positions, but it feels like I could streamline this process with the tool you’re working on!

closed 2021-08-17 by joeycastillo