joeycastillo/Sensor-Watch #4: Button release?

jamesbowman, 2021-08-22:
I'm wanting to duplicate the original behavior with the LED. Specifically the LED comes on while BTN_LIGHT is pressed.

However I don't see a button release event? That is, doing this:

    watch_register_button_callback(BTN_LIGHT, cb_light_pressed);

means that `cb_light_pressed` is pressed once when the button goes down.

Could the callback get an argument, 1 for pressed and 0 for released?

gennyble, 2021-08-22:
I don't believe adding an argument for pressed/released would work here. The buttons operate using interrupts which I think is activated when the pin is powered or unpowered, but not both. To have the callback take an argument would require the watch to poll the button every tick. I linked a section of code from Joey's BME280 application below. It'd be real useful to know when a button is released, so I hope I'm wrong.

https://github.com/joeycastillo/Sensor-Watch/blob/main/Sensor%20Watch%20BME280%20Project/app.c#L106-L114

joeycastillo, 2021-08-23:
The EIC can fire on rising, falling, or both edges, but the low-power RTC tamper interrupt (aka "external wake") that currently controls the alarm button can only fire on one or the other. 

Having said that, I think this is a useful feature to have. I think perhaps I should refactor the `register_button_callback` function into two functions: `register_eic_callback` that can provide both rising and falling indications for any EIC pin, and a separate `register_extwake_callback` function for folks that want to use the alarm button without enabling the EIC (see #5). 

Will look into this.