joeycastillo/Sensor-Watch #5: Possibly ~8% (~5µA) Power Savings By Disabling The EIC

gennyble, 2021-08-22:
Looking at the docs, `watch_enable_buttons()` has this note attached:

> The BTN_ALARM button runs off of an interrupt in the the RTC controller, not the EIC. If your application ONLY makes use of the alarm button, you do not need to call this method; you can save ~5µA by leaving the EIC disabled and only registering a callback for BTN_ALARM. 

I wonder if we can disable the EIC when we don't need it. I'll describe the strategy I'm trying in my application below. I've implemented it successfully, but I can't measure current as low as this is so I don't know if the EIC is *actually* powered down.

The watch is idle most of the time which probably means we don't need all the buttons to work. In my app I have a 5 second timeout where if no buttons were pressed, the EIC is deinitialized ([here][deinit]) and reinitialized when the alarm button is preesed ([here][reinit]). There's a 250ms green LED flash on deinit and reinit that tells the user it's gone into/come out of idle.

[deinit]: https://github.com/gennyble/neralie-sensor-watch/blob/main/app.c#L119-L129
[reinit]: https://github.com/gennyble/neralie-sensor-watch/blob/main/app.c#L95-L108

gennyble, 2021-08-22:
I looked a bit further in to the code and the SAML22's docs, and it seems it's possible to disable and enable the EIC without having to reinitialize. I ripped some code out of `hpl_eic.c` ([here](https://github.com/joeycastillo/Sensor-Watch/blob/42f0eac6d12bb354ce34f1cee42ad43f8a023282/watch-library/hpl/eic/hpl_eic.c#L155-L158) and [here](https://github.com/joeycastillo/Sensor-Watch/blob/42f0eac6d12bb354ce34f1cee42ad43f8a023282/watch-library/hpl/eic/hpl_eic.c#L170-L174)) and have it working as demonstrated [here](https://github.com/gennyble/neralie-sensor-watch/blob/e3730bdf09e0b594e7c1b85e995527c8e8c99d95/app.c#L50-L65).

joeycastillo, 2021-08-23:
I would definitely like to add a function to the library disable the EIC. There are also a couple of other functions missing that would disable e.g. the segment LCD controller or the ADC; I just haven't gotten around to them. 

If you added this function to the library I would totally merge it in; otherwise I will take a crack at it later this week.