joeycastillo/Sensor-Watch #6: Unable to run make

levimaaia, 2021-08-23:
Following the quick start guide included with the board on my iMac, I'm unable to get `make && make install` to run and write code to the board. For example, I `cd` to the `Sensor Watch BME280 Project/make` directory and enter `make && make install` and it kicks back the following three lines.

```
CC build/app.o
make: arm-none-eabi-gcc: No such file or directory
make: *** [build/app.o] Error 1
```

I did run `brew install gcc-arm-none-eabi` to install what I think is the right support for the board. Am I missing something else?

joeycastillo, 2021-08-23:
Hmm! The command to install this is `brew install --cask gcc-arm-embedded`, but it looks like this now installs the `10-2020-q4-major` version, whereas the watch project requires `9-2019-q4-major`. 

You can [download the toolchain here](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads/9-2019-q4-major) and put the tools in your path; alternatively, you can download [this previous version of the formula](nano -w /usr/local/Homebrew/Library/Taps/homebrew/homebrew-cask/Casks/gcc-arm-embedded.rb), and drop it in in place of the formula in `/usr/local/Homebrew/Library/Taps/homebrew/homebrew-cask/Casks/gcc-arm-embedded.rb`. Then you can run `brew install --cask gcc-arm-embedded` and it will install the correct version. 

This is a definite friction point and I'll need to figure out a way to simplify and document this process.

levimaaia, 2021-08-23:
Thanks for the quick reply! I tried the ladder and it seems to work. At least I'm past that error and on to new ones. These look more local to me though. Since I'm in the process of updating from macOS 10 to 11.5.2 and updating XCode, I'll work through that before I bug you again.

joeycastillo, 2021-09-15:
I had planned to leave this open until I better documented installing the old version, but instead, good news: the watch library now works with the latest `gcc-arm-embedded`! Will update the readme to remove the 2019-q4 note.

closed 2021-09-15 by joeycastillo