joeycastillo/Sensor-Watch #13: App contributions?

tahnok, 2021-09-14:
I've finished writing my first app for the sensor watch, witch displays the current Swatch Internet time. You can check out the code over here (https://github.com/tahnok/Sensor-Watch/commit/bd4626ab2f1af79674947f793fff01268f459724) but before I open an PR with it, I'm curious to know you want to handle contributed apps / firmware.

I quite like the way that the [Goodwatch](https://github.com/travisgoodspeed/goodwatch/tree/master/firmware) handles this, which that there is a single firmware codebase. A user can configure what functionality they want at build time by changing a Makefile. App writers don't have to worry about code for the 'core' functionality of watch (like setting the time).

Basically, I would love if there was a way to have firmware with some normal watch functionality (like a stopwatch or a countdown timer) and some weird watch functionality (like a random number generator or weird time keeping systems) combined into a single codebase somehow, while still allowing someone to toss it all and use the (wonderful) framework you are putting together

joeycastillo, 2021-09-15:
This is awesome!! I totally agree that a system like what's offered by the Goodwatch would be ideal, but I haven't finished thinking through the UI for it. Whereas the Goodwatch has a button matrix as well as a side button, the F91-W only has three buttons; claiming the Mode button for switching apps only leaves two buttons for apps to work with. Perhaps a two-button gesture or a press-and-hold could move between apps? Either way, I think that whatever "launcher" app ends up making sense will be one level of abstraction up from this library; that is to say, it will probably be another project at the root of the repo, and apps that want to be a part of it will go inside of that project and interact with a more high-level API. 

For now, though, I think that there should probably be an 'apps' folder for apps like yours that work with the low-level library directly, but I think I need to do a Makefile tweak to make that make sense. I will update this issue when I make that change (tho it may require a small change to your project's Makefile).

joeycastillo, 2021-09-16:
OK! I've rearranged the repo slightly: there is now an "apps" directory, and your project can go in there. 

The Makefile changes required are small and you can see them [here](https://github.com/joeycastillo/Sensor-Watch/blob/main/apps/Sensor%20Watch%20BME280%20Project/make/Makefile). In short: 

* add `TOP = ../../..` at the beginning
* change the make.mk line to `include $(TOP)/make.mk`
* change the rules.mk line to `include $(TOP)/rules.mk`

tahnok, 2021-09-25:
Yeah, the 3 buttons certainly poses a user interface challenge. I think a long press would be easier to use then a double tap, but I think a more complex 'button input handling + callback' library would be useful. I'd like to take a crack at that, but I might not get to it for a week or 2. My dream is to have a quick and easy shortcut from the main watchface to start a 3min tea timer

I've applied the Makefile changes you've suggested and opened up a PR: https://github.com/joeycastillo/Sensor-Watch/pull/15

joeycastillo, 2021-10-04:
Renamed this issue to focus on what you mentioned in your initial comment (firmware with some normal watch functionality and some weird watch functionality). I've been working on something like that [in this branch](https://github.com/joeycastillo/Sensor-Watch/compare/wip-launcher); it's not ready to merge and the nomenclature / API is likely to change, but figured I'd put it forward as a source for ideas. 

The gist is that the "launcher" is a bare metal app much like the apps we've been writing so far, and it manages the display of a number of "widgets" that can take over the display, one at a time. Each widget has:

* a setup function (called at boot and wake from sleep)
* an activate function (called before coming on screen)
* a loop function, which is also passed an event (like a button press) that caused the function to be called
* a resign function (called before going off-screen)

I made a few widgets just to test out the system (clock, preferences, and set time), and it seems like it might be workable. This may not be as powerful or customizable as someone rolling their own app from scratch, but could be a starting point for folks who want to add one interesting screen without having to worry about core functions like setting the time. 

Open to thoughts and comments from anyone!