joeycastillo/Sensor-Watch #14: Build PCB

elalemanyo, 2021-09-20:
Hi,
Maybe this is a rookie question, but how can I get/build the PCB?

Thanks

joeycastillo, 2021-09-20:
I'm working on having a run of 100 manufactured in the near future; that will be a way to get the fully assembled gadget. 

You can order the PCB yourself using the gerbers; they're in the `OSO-SWAT-A1-05-BLUE_2021-09-20.zip` file in [PCB/Main Boards](https://github.com/joeycastillo/Sensor-Watch/tree/main/PCB/Main%20Boards). Note that if you do this, you will need to order the PCB in 0.6 mm thickness for it to fit in the watch case. 

Still, I would recommend waiting for the fully assembled boards; the ATSAML22J18A-MUT that I'm using is out of stock basically everywhere right now, so you would have a hard time building it on your own right now regardless.

closed 2021-09-20 by joeycastillo

tangbao, 2021-09-29:
Hi,

It is exciting to know that I can buy fully assembled boards in the future!!

For ATSAML22J18A-MUT, have you ever tried to find sellers from China (like AliExpress)? For example, you can buy 10 pcs for $37.8 at https://www.aliexpress.com/item/4000775253513.html. I believe that it will be cheaper if you buy 100 or more. Some other things like resistances, sensors, and PCB manufacturing are cheaper in China as well. The only thing we need to worry about is the shipping time, which could take one month or more (but the shipping is cheap as well).

I'm glad to help you if you think it could be a good idea (I'm not sure) to contact AliExpress sellers / PCB manufacturers in Chinese.

Thank you for your great project! I cannot wait to get my own sensor watch!

tangbao, 2021-10-08:
Would you please consider this suggestion? @joeycastillo

tangbao, 2021-10-11:
> > Would you please consider this suggestion? @joeycastillo
> 
> I am currently on track to produce a first run of 100 boards domestically, using chips sourced from Mouser before the shortage. I am not as certain about sourcing chips from Aliexpress; either way, right now my focus is on getting this run produced and in folks’ hands before considering anything like that.

Yes, I fully understand that. I thought you did not have enough chips to make 100 boards. I was just providing some suggestions on where to buy chips and other supplies.

Thank you for your work and looking forward to its publish day!