joeycastillo/Sensor-Watch #22: MIgrate hathach/tinyusb to a Git submodule

willianpaixao, 2021-11-11:
Having the [hathach/tinyusb](https://github.com/hathach/tinyusb) code checked out at this repo means you miss bug fixes and future features, or the burden of constantly updating the code.

I suggest removing the static code and adding back as a [Git submodule](https://git-scm.com/book/en/v2/Git-Tools-Submodules), this is the proper way to add other repositories as dependencies to your code. Updates would be done semi-automatically and new clones would always bring the newest version.

### References
- https://github.blog/2016-02-01-working-with-submodules/
- https://www.atlassian.com/git/tutorials/git-submodule

joeycastillo, 2021-11-11:
I sincerely considered this when setting up the dependency. I opted to include the static code because I sense that git submodules can be confusing for the beginner, and I wanted to make it as simple as possible for folks to check out the code and get building quickly.

Leaving the issue open for comments because while I absolutely agree that including the code as a submodule is the correct way to do it, I’m trying to balance that with the desire to make it easy for folks who aren’t as well-versed in git. If there were multiple dependencies I could see it, but the burden of occasionally pulling updates for this one module feels like a worthwhile tradeoff to make the onboarding of new folks as smooth as possible.

tahnok, 2021-11-11:
I ran into a similar conundrum while working on a TOTP face, I needed to
include https://github.com/Netthaw/TOTP-MCU and I opted to do that via
submodule but was going to bring that up for discussion. My thought was
that not everyone will be building the TOTP so it could be an 'optional'
dependency. I also wasn't sure how to properly credit the code if I were to
vendor it. On the other hand it's only 4 files and I'm not sure how often
it gets updated so vendoring it might be worth it.

I will add that submodules suckkkkkkkkkkkkk. I've been using git for 10
years and I still need to look up all the weird commands for submodules

joeycastillo, 2021-11-11:
Interesting data point: right after making this comment, I went to rebuild [the UF2 bootloader for the watch](https://github.com/joeycastillo/uf2-samdx1) and noticed that the make command for that project automatically pulls in the required submodules if they’re not available. 

I didn’t look too deeply into how it’s implemented, but if we could make this bit automatic and part of the build process, that might be a way to adopt submodules while not adopting the headache of making folks deal with them.

PR #25 (2021-12-08): Add tinyusb as git submodules
Fixes #22
closed 2021-12-08 by joeycastillo