joeycastillo/Sensor-Watch #42: Small updates to Movement README

davidskeck, 2022-01-20:
Hello,

I would like to suggest a few updates to the Movement README.

1. Consider adding a section that details the need to add the new watch face file to the Makefile.
2. Consider explaining the purpose of the Boolean return in watch_face_loop.

These two items tripped me up when making my own watch face.

Thanks!

joeycastillo, 2022-01-25:
Thanks for the feedback! Definitely agree on no. 2, I need to do a better job explaining that. 

As for no. 1, I think the next step for movement is to try to automate some of this. Ideally I'd like to have like a Python script that one can run something like `watch-face new tide_chart`, and have it generate the header and implementation file with boilerplate, and handle the appropriate includes and Makefile additions. This would naturally involve a bigger rewrite of the README to accommodate the new process, but I'd like too get started on this soon so I might put a pin in this issue and address all README tasks when that's done.

tahnok, 2022-01-25:
In the short term / low effort, you could link to a PR that adds a watch
face as an example for people to follow

davidskeck, 2022-01-25:
> Thanks for the feedback! Definitely agree on no. 2, I need to do a better job explaining that.
>
> As for no. 1, I think the next step for movement is to try to automate some of this. Ideally I'd like [...]

That sounds great. Would you like any help making that script? Agree that linking to a PR short term would be helpful.

joeycastillo, 2022-01-25:
> Would you like any help making that script?

I would welcome that help! I sense I would get to this eventually but if it's something you're interested in, by all means, I'd be stoked if someone wanted to take a crack at it!

The conversation around this issue lit the fire under me to [add a template](https://github.com/joeycastillo/Sensor-Watch/commit/f819a18791d6ad5a4f68125928587d0f87833a1c) and slightly reorganize the way we're doing watch faces (moved the lengthy list of `#include`s to a separate file, and renamed a couple of folders). Realizing now that the folder structure means we need two parameters, the type of watch face and the name (i.e. `watch-face new complication tide_chart` or `watch-face new clock mars_clock`. 

And then short-term, I think I can link to [this diff](https://github.com/joeycastillo/Sensor-Watch/commit/2a59ae950f653a1730686ede8f77d74aea125efe) that's a really clean addition of a single watch face. I'll do that shortly.

perigoso, 2022-02-14:
This is something I can help with too, Make can be cumbersome, perhaps something like meson would help keep the build system clean and dynamic, I can help with that, It's not ideal for embbeded systems but I work with it and it's still better in a couple ways than the other options imo

joeycastillo, 2023-01-11:
Closing this issue, as documentation was added in #42 and the watch face generation script in #47.

closed 2023-01-11 by joeycastillo