joeycastillo/Sensor-Watch #71: Stange behavior with side light

davidskeck, 2022-05-18:
Hello,

I was using my watch, and attempted to use the side light and got some odd behavior.

For context, I did have the light set to mix the green and red colors to make a somewhat yellow color. This had worked previously, but I had noticed it was a little slower to respond, and then the attached video occurred. I was able to stop this but only by going into settings and turning off the red portion of the light.

Any ideas? I have since reflashed my board (to get the new tomato face!)

https://user-images.githubusercontent.com/1740630/168938319-c504408b-673a-4f4b-a52f-2a2dafda4e32.mp4

joeycastillo, 2022-05-18:
interesting! I would expect the light to get stuck on red when the watch enters standby mode, as the PWM driver stops running in standby. But Movement should prevent the watch from entering standby for as long as the LED is on. 

My guess is that this is a glitch with the `light_ticks` property in movement_state; I'll see if I can replicate it.

davidskeck, 2022-05-19:
I wanted to mention that when I disassembled the watch after this occurred I noticed that the metal tab to press the light button was flaky due to it being bent. I replaced it and it works well now, but it could have been hammering the contact multiple times a press if it was making intermittent contact.