joeycastillo/Sensor-Watch #83: Bootloader

jackbondpreston, 2022-08-23:
Is the bootloader for the sensor watch available? What's the process for flashing from scratch?

Apologies if it's somewhere obvious.

joeycastillo, 2022-08-23:
No need to apologize, it is in fact not anyplace obvious. [Here's the link](https://github.com/joeycastillo/uf2-samdx1); it's a fork of Adafruit's UF2 bootloader with support for the SAM L21 and SAM L22. 

The process for flashing from scratch involves connecting to the five test points on the front of the Sensor Watch board, either by soldering to them or using pogo pins. You can use a PyRuler or Adafruit Trinket and [this Arduino sketch](https://github.com/joeycastillo/Sensor-Watch/tree/main/utils/flash_watch_pyruler) to flash a precompiled bootloader to the board; this will also set the BOOTPROT fuses to prevent the bootloader from getting overwritten accidentally. (the sketch also requires the [Adafruit_DAP](https://github.com/adafruit/Adafruit_DAP) Arduino library.

Hope this helps! Also I'm going to leave this issue open because I do need to put this documentation somewhere so it's easier to find; I will close it once that's in place :)

jackbondpreston, 2022-08-23:
Thanks so much, you are eternally helpful!

joeycastillo, 2022-08-28:
You are so right, I did have to make that change to Adafruit_DAP, and we should endeavor to upstream that support.