joeycastillo/Sensor-Watch #127: [Security] Manual suggests to put TOTP on the web

bademux, 2022-11-09:
Sorry for this issue, I'm sure we all grownups and can  handle our secrets, but...

Please don't direct people with private things like TOTP to the web sites. It simply bad idea to teach bad behavior on the web.

The manual twice suggest to to the web https://www.sensorwatch.net/docs/watchfaces/complication/#totp-generator
1. [Stefan Sundin’s web site](https://stefansundin.github.io/2fa-qr/)
2. [cryptii.com](https://cryptii.com/pipes/base32-to-hex)

For the TOTP secrets KeePssXC local application can be used.
converting base32 to C style bytes as simple as:
```bash 
$ echo "KRUGKIDROVUWG2ZAMJZG653OEBTG66BANJ2W24DTEBXXMZLSEB2GQZJANRQXU6JAMRXWOLQ=" | base32 -d | xxd -i
  0x54, 0x68, 0x65, 0x20, 0x71, 0x75, 0x69, 0x63, 0x6b, 0x20, 0x62, 0x72,
  0x6f, 0x77, 0x6e, 0x20, 0x66, 0x6f, 0x78, 0x20, 0x6a, 0x75, 0x6d, 0x70,
  0x73, 0x20, 0x6f, 0x76, 0x65, 0x72, 0x20, 0x74, 0x68, 0x65, 0x20, 0x6c,
  0x61, 0x7a, 0x79, 0x20, 0x64, 0x6f, 0x67, 0x2e

```

tahnok, 2022-11-10:
Nice, I like the idea of telling people how to do this with a CLI, but I don't see the problem using webtools for someone who's not technical. The risk of one of these sites somehow stealing your TOTP secret and actually being able to do something with it seems pretty unlikely

Are you open to putting a PR up on https://github.com/joeycastillo/sensorwatch.net/blob/main/content/en/docs/watchfaces/complication.md ?

bademux, 2022-11-10:
@tahnok yep, I will add PR.
Unfortunately I can't agree with about web tools. We take responsibility of teaching people bad practices.
Combined with social hacking it will be fairy simple to take over someone's access.

closed 2022-11-10 by bademux