joeycastillo/Sensor-Watch #132: Inconsistent bell/signal indicator usage

wryun, 2022-11-15:
Previously, faces (e.g. wake/countdown/tomato) used the `WATCH_INDICATOR_BELL` to say that there was an alarm coming up. The cool new alarm face (and the 'alarm_enabled' bit of simple_clock) now use `WATCH_INDICATOR_SIGNAL`.

I'm guessing that those faces should be changed to use `WATCH_INDICATOR_SIGNAL`.

On the other hand, I'm not sure I like the change. The thermistor/voltage/accelerometer faces use `WATCH_INDICATOR_SIGNAL` when they read data, and this fits better than the bell IMO. i.e. we're now overloading SIGNAL, and under-using BELL (which will be left with only the hourly chime?).

Thoughts - @TheOnePerson , @joeycastillo ?

joeycastillo, 2022-11-15:
For the hourly signal, this was originally a mistake of mine: the usage on the original F-91W is that the bell icon indicates the status of the hourly signal, and the signal icon indicates the status of the alarm. To make it feel consistent for folks, we swapped them to match. 

I'm of the opinion that watch faces should feel free to use whatever indicator makes the most sense for their UI: 

* Wake should probably switch over to signal, since it is an alarm.
* In the case of sensor faces, I think the signal icon also makes sense; it feels to me like data being sampled. 
* In the case of countdown/tomato, I'm agnostic about which one is used. Signal would match the "alarm"iness of the main face, but the bell could indicate a warning or alert condition, like “Be vigilant, you have cookies in the oven”.

I think it's fine to underuse bell, especially if lots of watch faces do use it in that manner. Signal is kind of a versatile indicator that's useful in a lot of places, whereas bell is more specific.

joeycastillo, 2023-01-11:
I'm going to close this issue, as the transition from signal to bell for hourly chime seems to have gone over okay with folks. If there are other areas where indicator selection is causing confusion, feel free to reopen or start a discussion in the Discord. 

Thanks for the feedback!

closed 2023-01-11 by joeycastillo