joeycastillo/Sensor-Watch #145: Are the BOMs for the sensor boards?

danielzfranklin, 2022-11-25:
I couldn't find any sort of BOM for the sensor boards. Am I missing something, or is what specific component to use something I should figure out myself?

(I'm mainly interested in the hiking log board.)

tahnok, 2022-11-27:
I don't believe there are BOMs, you'll have to look at the schematic and
make your own (or maybe eagle can export one?)

jackbondpreston, 2022-11-27:
Eagle can indeed export BOMs. For example, for the hiking log:
```
Partlist exported from C:/Users/jackb/Sensor-Watch/PCB/Sensor Boards/OSO-MISC-21-008 Hiking Log.sch at 27/11/2022 23:50

Part Value                 Device                Package                Description                                                                                    DESCRIPTION                                                              DIGI-KEY_PART_NUMBER DIGI-KEY_PURCHASE_URL                                                                                                                                     MF                 MP       PACKAGE                  
C1   10µF                  CAP_CERAMIC_0402MP    _0402MP                Ceramic Capacitors                                                                                                                                                                                                                                                                                                                                                                                          
C2   0.1uF                 CAP_CERAMIC_0402MP    _0402MP                Ceramic Capacitors                                                                                                                                                                                                                                                                                                                                                                                          
R1   1% 10K                RESISTOR_0402MP       _0402MP                Resistors                                                                                                                                                                                                                                                                                                                                                                                                   
RT1  NTC 10K               RESISTOR_0402MP       _0402MP                Resistors                                                                                                                                                                                                                                                                                                                                                                                                   
U$3  FH19C-9S-0.5SH_10-FFC FH19C-9S-0.5SH_10-FFC FH19C9S05SH10-FFC                                                                                                                                                                                                                                                                                                                                                                                                                  
U1   LIS2DHTR              LIS2DHTR              PQFN35P200X200X100-14N MEMS digital output motion sensor: ultra low-power high performance 3-axes femto accelerometer  Accelerometer X, Y, Z Axis ±2g, 4g, 8g, 16g 0.5Hz ~ 672Hz 14-LGA _2x2_  497-13795-1-ND       https://www.digikey.com/product-detail/en/stmicroelectronics/LIS2DHTR/497-13795-1-ND/4209308?utm_source=snapeda&utm_medium=aggregator&utm_campaign=symbol STMicroelectronics LIS2DHTR LGA-14 STMicroelectronics
```

note that `FH19C-9S-0.5SH_10-FFC` is not something you need as this is the watch-side connector for the sensor PCBs. so the only specific component you need is the `LIS2DHTR`, which you can google (or use the DigiKey link provided). the rest are fairly generic components, as long as you buy something fitting the specified package size, resistance/capacitance, etc.

joeycastillo, 2022-11-28:
So FYI, I haven't tested all of the boards in that folder. The project kind of migrated from "thing I'm hacking on" to "product that people use" without me doing a pass to clean up a lot of sort of draft work. Case in point: I never actually tested the Hiking Log board, as the LIS2DH was way too fine-pitched for me to place consistently. Instead I've moved on to using the LIS2DW. That accelerometer does technically have a temperature sensor inside, tho I haven't written code to make use of it yet or gotten a sense for its accuracy. 

There's a tested design called Sensor Watch Motion II in there (OSO-MISC-21-021), and the BOM for that is just the LIS2DW and two 0603 capacitors (10µF and 100nF).