joeycastillo/Sensor-Watch #150: [Documentation] Debugging/simulating LFS

v-p-b, 2022-12-10:
Related to #62 

I was trying to get the LSF-based TOTP face to work, but couldn't make it parse the .txt file. It would be nice to know how to place files on the LFS of the simulator, so issues related to file format can be debugged more easily.

wryun, 2022-12-11:
It's impossible (AFAIK) to use a serial connection to put files on the watch due to https://github.com/joeycastillo/Sensor-Watch/issues/117

I've put TOTP URLs on my watch by changing the firmware to write the files to LFS (i.e. hard-code in firmware), then replacing the firmware with the normal one. Yuk :(

The simulator has a pretend serial connection in the visible console, so you should be able to use echo ... >> totp_uris.txt as documented in https://github.com/joeycastillo/Sensor-Watch/blob/main/movement/watch_faces/complication/totp_face_lfs.c

If you then move to the totp face, any errors it has while parsing the file should come out on the same console.

v-p-b, 2022-12-11:
@wryun Oh OK, thanks for the heads-up, so it wasn't my URL's!

Also, the second part of your comment answers my question, so I'm closing this.

closed 2022-12-11 by v-p-b