joeycastillo/Sensor-Watch #156: GPS Sensor watch accessory board

Asalle, 2022-12-27:
First of all, thanks for this amazing project, it's high-key genius!

I wanted to ask if it's realistic to add a gps module, save gps data and offload it to a PC via usb(or somehow else)?

joeycastillo, 2022-12-28:
Hey and thanks for the kind words!

Unfortunately, a GPS module is unlikely to be a good fit for Sensor Watch. The antennas required for GPS reception are generally more bulky than could fit in the Casio watch enclosure, and even the most power efficient modules consume double digit milliamperes when getting a fix. For a device that has only 100 milliampere hours of energy to work with, this would kill the battery in a matter of days, instead of the year of battery life we get by sticking to the microampere range. 

Hope this helps and sorry if it's not the answer you were hoping for!

closed 2022-12-28 by joeycastillo

Hein-NonesensE, 2023-01-22:
Just for the record, as I had the same thought and looked into it: I've found small package GPS sensors by u-blox that might just fit the watch, but only with a custom antenna. Some of these can operate in the single digit milliampere range in 1Hz power saving mode (don't know if that mode works right after cold start). Might just work when deactivated completely while GPS face not active, but requires a lot of work (as I said: Custom antenna etc). Maybe in a few years we'll see modules that are small and economical enough to simply drop in.