joeycastillo/Sensor-Watch #166: movement_illuminate_led() mangles EVENT_*_LONG_PRESS behavior

enthdegree, 2023-01-09:
`EVENT_LIGHT_BUTTON_UP` shouldn't happen after `EVENT_LIGHT_LONG_PRESS`. 
However, if you call movement_illuminate_led() during an `EVENT_LIGHT_LONG_PRESS` then `EVENT_LIGHT_BUTTON_UP` happens once the button is released.

Here is an example loop:

````
 bool test_face_loop(movement_event_t event, movement_settings_t *settings, void *context) {
    (void) settings;
    (void) context;
    switch(event.event_type) {
    case EVENT_MODE_LONG_PRESS:
        movement_move_to_next_face();
        break;
    case EVENT_LIGHT_LONG_PRESS:
        movement_illuminate_led(); // If you comment out this line everything works properly
        watch_display_string("long", 4);
        break;
    case EVENT_LIGHT_BUTTON_UP:
        watch_display_string("short", 4);
        break;
    }
}
````

TheOnePerson, 2023-01-11:
I think I found to source of this behavior: In `movement.c`, the function `movement_illuminate_led()` calls `_movement_enable_fast_tick_if_needed()`, which looks like this:
```
133    static inline void _movement_enable_fast_tick_if_needed(void) {
134        if (!movement_state.fast_tick_enabled) {
135            movement_state.fast_ticks = 0;
136            watch_rtc_register_periodic_callback(cb_fast_tick, 128);
137        }
138    }
```

The problem here is, that `_movement_enable_fast_tick_if_needed()` does not set `movement_state.fast_tick_enabled = true`. So each time this function is called, the fast_ticks counter is reset so zero. Since both the button long press timer *and* the timer for the led rely on fast_ticks, the timing for long pressing of a button gets messed up when you initiate the led.

The solution should be trivial: Just add a `movement_state.fast_tick_enabled = true` at line 135 and we should be good to go.

Could you perhaps test this with your build?

enthdegree, 2023-01-18:
Thank you for tracking this down @TheOnePerson . My test face has the expected behavior on your branch in #183 with the latest merged in from joeycastillo/Sensor-Watch:main ([5163cf1](https://github.com/joeycastillo/Sensor-Watch/commit/5163cf1fe1e8bf878bf8d4e54b1fa24dccb4e116)).

enthdegree, 2023-01-25:
I think that merge resolved it.

closed 2023-01-25 by enthdegree