joeycastillo/Sensor-Watch #197: [Feature Request] Set clock time as a mode of the clock, instead of it's own face

strickinato, 2023-01-29:
This is something that always bothered me about the stock f91w interface, and since we now are in control, it would be lovely to fix:

### **The problem: It's annoying that setting the time is it's own mode**

**Situation 1:** 
I have a number of faces that I really love. I cycle through them, and then all the sudden, I'm in the danger zone of changing the time!! My excitement and flow is broken.

**Situation 2:**
In the middle of the night, I futz around with my watch. It's dark. Where am I? I may need to wake the watch up with the "alarm set" button, but... if I'm on the wrong face, that might unwantedly set the time!

### **The potential solution:**

The `set_time_face` is _**not implemented as it's own face**_, but instead is accessed via a long press of the mode button while on the `simple_clock_face`!

**Why I think this would be an improvement:**
1. You always want to have the `set_time_face` if you have the `simple_clock_face`. They should go together
2. long pressing the `mode` button is not utilized in `simple_clock_face`.
3. You don't end up with the annoying problems listed above

TBH - I'm not sure if this is difficult or not to implement, but if perhaps if the community feels how I feel, someone could take on this improvement! Or if people are really excited, but no one wants to take it on, I can try my luck at an implementation.

<3 Thanks again for this project, I love my watch so much!

neutralinsomniac, 2023-01-29:
Are you aware of `MOVEMENT_SECONDARY_FACE_INDEX` in `movement/movement.h`? It can be used to separate settings-based or non-frequent watch faces into a secondary face rotation that is only accessible by long-pressing mode from the first watch face.

strickinato, 2023-01-29:
Incredible!! I did not know about it - but that looks perfect. Thank you 🙏

closed 2023-01-29 by strickinato