joeycastillo/Sensor-Watch #201: Watch locked/frozen up on interval face

kingannoy, 2023-02-01:
My watch locked up today. 

I asked for help on the [discord](https://discord.com/channels/969627287429459988/1070303222297792573) and was asked to write a bug report, so the problem could be tracked. I'm going to put in all the information that I think might be relevant.

It got in this state when I long pressed the alarm button on the interval face. I think I was on the 5th default alarm, with the 25/5 pomodoro timer. This was about 10 hours ago, and it's currently still in that state.

It is only showing the starting time of the break period on the interval face. Not counting down, no way to move away from there. This was the first time I was attempting to use this face, so I'm unsure if this is a fluke. I'll open up the watch and see if I can reproduce if I restart the watch without changing the code. I'll update when I know more.

I think this is the relevant bit of my `movement_config.h`:

```
const watch_face_t watch_faces[] = {
    simple_clock_face,
    alarm_face,
    stopwatch_face,
    interval_face,
    countdown_face,
    totp_face,
    day_one_face,
    thermistor_logging_face,
    thermistor_readout_face,
    weeknumber_clock_face,
    preferences_face,
    set_time_face,
};
```

It was built on a friends computer on 21 Jan.

I've not been able to reproduce this behavior in the emulator

Now that I've played with the emulator (and understand better what the interval face is supposed to do) I think it is interesting that it's locked up on the starting time of the "break" period. So not on the "work" period where it is supposed to start. I'm including a photo because think the state it got stuck in might contain a hint for someone that knows the code :sweat_smile: 

[image]

kingannoy, 2023-02-01:
I have not been able to reproduce the behavior after a restart, so it seems like I was just unlucky somehow.

I also think I was wrong. In the emulator the interval I used was the fifth one. I think my friend may have changed this, or the interval face defaults may have changed in the past month. But on my watch the first timer is set to be the pomodoro (I've not changed the timers from how they were set in the image).

I've included some photo's of the settings I see on the interval that I think triggered the lock-up, in the order in which they show in the carousel:
[image]
[image]
[image]

That explains why the top and bottom right numbers are 1 and 4, because that is also how they show in the carousel. So not as much of a hint as I had thought or hoped...

TheOnePerson, 2023-02-04:
Oh, this look really odd, esp. the last picture showing the cool-down phase. First of all, there is no regular way of setting this to 22 seconds, since the settings mode counts in steps of five. Secondly the lower right hand corner is supposed to be blank - I wonder where the 6 comes from? It looks to me like the internal settings state has been messed up. By what? No clue so far...

TheOnePerson, 2023-02-04:
You mentioned on discord that your friend has made some (minor) modifications to the face's code for you. Could you send me the version of interval_face.c that you actually used for your watch?

kingannoy, 2023-02-21:
I wish I could, but my friend is not sure he still has this file, and hasn't found the time to search for it yet. So I think based on the pictures not looking like they should, we can close this issue and say that it was probably a bug introduced by the editing of the face.

closed 2023-02-21 by kingannoy