joeycastillo/Sensor-Watch #203: Temperature sensor not reporting temperature properly

digi-chris, 2023-02-02:
Hi,

Hope this is the right place to post this!

I have four sensor watch boards and four of the temperature sensor breakout boards.

After (finally!) putting them all together, three of the watches show the correct temperature. The fourth is showing a temperature usually around -70 degrees Celcius. I've tried reseating the connections a couple of times to see if that's the problem, but it's not making any difference.

Is there any further debugging I can try to do to fix the problem? Or, it is likely the sensor is malfunctioning and will need replacing? Happy to buy another if so.

Thanks!

Chris.

Jonas-fr, 2023-02-02:
Here's a suggestion : label your watches from W1, W2, W3 and W4. Lablet your sensors S1, S2, S3 and S4.

Right now you're in this situation :
W1+S1 : OK
W1+S1 : OK
W1+S1 : OK

digi-chris, 2023-02-02:
Thanks @Jonas-fr ! I don't know why I didn't think of that, but that makes perfect sense.

When I get a chance, I'll swap two of them over - will likely be tonight or tomorrow.