joeycastillo/Sensor-Watch #242: Clock running too fast

EmiliaBlasten, 2023-04-21:
My watch seems to gain about 30 seconds every couple of months. This makes the 2-factor authentication watch face useless unless I reset the time once or twice a month. The gain seems to be consistent but I haven't measured it. 

Do others have the same issue?
- If yes it's probably due to the code.
- If not then the issue is likely due to manufacturing tolerances of the quartz resonator in which case (if it's a consistent gain for each resonator) we should create a calibration watch face or code.

DefProc, 2023-04-21:
I used to see a higher drift than that, but there's a very accurate calibration procedure now: https://www.sensorwatch.net/docs/watchfaces/nanosec/

neutralinsomniac, 2023-04-21:
Have you flashed a recent version of the firmware? The frequency correction offset was changed within the last couple months to bring the watch closer to an acceptable average drift. You don't need the nanosec face unless you want extremely accurate precision.

EmiliaBlasten, 2023-04-21:
Thanks! The version I have flashed is more than a few months old. I will try the new version (and also look at the nanosec watch face for curiosity) and report if there's any noticeable drift in a few months.

EmiliaBlasten, 2023-05-06:
So 15 days ago (when I posted the last comment), I synchronized the watch with the clock on my phone. I used the old firmware till today. Today the watch is 24 seconds in the future. I will now pull and install the latest firmware.

814d3, 2023-05-10:
I experienced the same behavior, also with a newer firmware (Jan/Feb '23). But finetune and nanosec do a fantastic job. Try them out and your watch will be very precise compared to not using them.

EmiliaBlasten, 2023-06-25:
It's been a around 50 days after synchronizing time on the new firmware. It's 11 seconds late compared to the phone's time. That's good enough for using 2-factor authentication. I will try nanosec / finetune next, when I have time.

annuvin, 2024-11-06:
my watch ran far too fast also, investigation of the ceramic capacitors on the quartz oscillator revealed one had been placed incorrectly making it out of circuit. i had to resolder it, it's quite tiny so if this is your issue i wish you good luck. my watch is dead accurate now.

EmiliaBlasten, 2024-11-07:
Forgot to close earlier. The issue was solved by upgrading to newer firmware.

closed 2024-11-07 by EmiliaBlasten