joeycastillo/Sensor-Watch #284: LED always on

roguepullrequest, 2023-09-26:
Any compiled firmware for LITE pcb the LED just stays full YELLOW. Pressing the Light button does turn off the GREEN for the set duration. 

Am i missing some default value or something.

roguepullrequest, 2023-09-27:
I did use the color RED. Sadly the Green led has burnt out because it was on too long.

aronhegedus, 2023-09-27:
I hit this today, and `make COLOR=RED` worked for me. I have a red board.

roguepullrequest, 2023-10-05:
So i have use clean repo and still the same issue. I always make sure color is RED.
i even tried other machines to rebuild and still same issue. 

The LED both green/red stay on.

joeycastillo, 2023-10-05:
To be clear, you're running `make COLOR=RED` (with both COLOR and RED in all capital letters)?

roguepullrequest, 2023-10-05:
> To be clear, you're running `make COLOR=RED` (with both COLOR and RED in all capital letters)?

Yup. Had to triple check my shell history to be sure but using `make COLOR=RED` to build.

Thanks for everything.

h6y3, 2023-10-07:
> > To be clear, you're running `make COLOR=RED` (with both COLOR and RED in all capital letters)?
> 
> Yup. Had to triple check my shell history to be sure but using `make COLOR=RED` to build.
> 
> Thanks for everything. 

I ran into this issue as well. I did a make clean before I issued the make command and that did the trick for me.

theorangerider, 2023-12-17:
Seems like this issue can be closed?

WesleyAC, 2024-01-21:
Going to close this as it seems to be the `COLOR` issue? @roguepullrequest if you're still having problems, definitely open a new issue.

closed 2024-01-21 by WesleyAC