joeycastillo/Sensor-Watch #286: TOTP Generation incorrect on Lite board

madhogs, 2023-09-27:
Hi,
The Totp generation on the sensor watch lite is giving incorrect values even for a test secret.

I have added just one test key here - https://github.com/joeycastillo/Sensor-Watch/compare/main...madhogs:Sensor-Watch:main
Then built the firmware using `make COLOR=RED`
Even after setting the time + date + time zone, the codes generated by the watch do not match the ones here https://totp.danhersam.com/
My own personal codes are incorrect too.

Its possible this is user error but i'm not sure what I could have done wrong with the test code in the above branch, and I have checked the datetime is correct.

tahnok, 2023-09-27:
> I have added just one test key here - main...madhogs:Sensor-Watch:main

Is your timezone set properly in settings?

madhogs, 2023-09-27:
Yes, I have it set to +1 which matches my timezone.
If I set the time to the current UTC time and timezone to 0 the codes are still incorrect.

Mr5quid, 2023-09-27:
Have you tried your firmware in the emulator ? It should rule out a time zone error since it's automatically set for the emulator.

I built my firmware for the Lite board with the latest sources on the 23rd, 4 days ago. TOTP generation is working just fine here.

madhogs, 2023-09-27:
It's wrong in the emulator too. I'm guessing i've done something wrong in the code above. I'm not what though, I used the https://cryptii.com page to convert the secret into hexadecimal bytes and added the 0x myself to each.

[image]

madhogs, 2023-09-27:
Sorry, I've just realised my mistake.

I've misunderstood the format of the secrets in the uri (and the test one even!). I assumed it was already decoded, I didn't realise it was base32 encoded. Decoding the secret before converting it to bytes (as the docs say!) and the codes generated are correct.

Sorry about this, it was user error after all. Thanks for your responses.

closed 2023-09-27 by madhogs