joeycastillo/Sensor-Watch #311: Upload firmware to the emulator

lchen198, 2023-10-31:
Is there a way to try my local build firmware in the emulator? Thanks

BrianBinFL, 2023-10-31:
This should probably be in "Discussion", not "Issue" - maybe someone with the proper rights can move it. Meanwhile I'll answer it.

You can read how to run the emulator locally [here](https://github.com/joeycastillo/Sensor-Watch/tree/main#using-the-movement-framework). Note that you'll need to install emscripten. There is a link on the page to the emscripten home. There are instructions for installing emscripten on various platforms on the Downloads page of the emscripten site.

I hope this helps.

WesleyAC, 2023-11-19:
Going to close this, since it seems like @BrianBinFL's link should answer the question. The reason that the firmware file can't be easily uploaded to the emulator is because that would require simulating the entire SAML22 chip, which is quite complex, while building the same source code with emscripten allows the simulator to work without needing accurate simulation of the hardware.

closed 2023-11-19 by WesleyAC