joeycastillo/Sensor-Watch #313: WATCHBOOT drive not showing up on Windows 10

stolksdorf, 2023-11-01:
Tried plugging my sensor watch board (red) into two different Windows 10 machines, and no additional disk drives popped up. Double tapped the reset button, the led was pulsing red, and tried both usb orientations just in case.

Can you only flash firmware via a unix machine?

joeycastillo, 2023-11-01:
It should work from Windows, Mac or Linux. Have you double checked that your USB cable isn’t a charge-only cable?

stolksdorf, 2023-11-01:
I tried two different cables, but third time's a charm! The cable was definitely the problem, thanks for the assist.

closed 2023-11-01 by stolksdorf

BrianBinFL, 2023-11-01:
> I tried two different cables, but third time's a charm! The cable was definitely the problem, thanks for the assist.

Also note that it matters which direction you face the board when you plug it in to the cable. You have to ensure that the side of the board with the contacts is facing the inside edge of the connector that has contacts. I've talked to at least one person who did not realize that.