joeycastillo/Sensor-Watch #327: Simple Watch Face Softlocked at Midnight w/ Hourly Chime Enabled

sntyj-dev, 2023-11-16:
I flashed firmware back on October 16th this year with only a few complications. I've also set the hourly chime to be the Zelda Tune. 

This morning, when I went to look at my watch it is stuck on midnight from today. It has the animation like it is still in standby, but I cannot wake it with the normal Alarm button press. 

This hasn't happened before, but the only thing that is different than normal is that I accidentally left the hourly chime on (typically I will keep that off, as I'm aware of issue #275  where the chime doesn't work in standby anyways.)

I have a few hours before I'm able to crack into the watch to pull the battery to reset. Is there something I should be dumping or checking?

WesleyAC, 2023-11-16:
If you wanted to give #325 a shot (it fixes the problem with the buzzer not working in LE mode), that would be useful — I'm curious if that would fix this or not, although it sounds like it isn't very reproducible, so it'd be hard to say for sure.

sntyj-dev, 2023-11-17:
Thank you for the suggestion. I've compiled new firmware from your fork, and will leave the hourly chime on overnight to see if it at least gets past midnight tonight. 

I have copied the same modifications from my previous build into the new build. Specifically the `movement_config.h` and `totp_face.c` files.

While I won't share what I'm doing in `totp_face.c` for obvious reasons, I will share the parts of  my `movement_config.h`  config that are different for reference just in case:
```

const watch_face_t watch_faces[] = {
    simple_clock_face,
    timer_face,
    world_clock2_face,
    sunrise_sunset_face,
    thermistor_readout_face,
    totp_face,
    time_left_face,
    rpn_calculator_face,
    voltage_face,
    preferences_face,
    set_time_face,
    finetune_face,
    nanosec_face
};

#define MOVEMENT_NUM_FACES (sizeof(watch_faces) / sizeof(watch_face_t))

/* Determines what face to go to from the first face on long press of the Mode button.
 * Also excludes these faces from the normal rotation.
 * In the default firmware, this lets you access temperature and battery voltage with a long press of Mode.
 * Some folks also like to use this to hide the preferences and time set faces from the normal rotation.
 * If you don't want any faces to be excluded, set this to 0 and a long Mode press will have no effect.
 */
#define MOVEMENT_SECONDARY_FACE_INDEX (MOVEMENT_NUM_FACES - 8) // or (0)

/* Custom hourly chime tune. Check movement_custom_signal_tunes.h for options */
#define SIGNAL_TUNE_ZELDA_SECRET
#include "movement_custom_signal_tunes.h"
```

sntyj-dev, 2023-11-17:
Update on this. My original issue has been resolved probably, as it made it past midnight without freezing with the chime on. Thanks again!

However, the chime itself still doesn't technically work as it should in LE mode. I can hear faint clicks when the buzzer is supposed to kick off, but the actual tune doesn't happen. Not sure if that's helpful @WesleyAC

sntyj-dev, 2023-11-18:
New build with proper branch and edits in place, the chime does work as expected! Even in LE mode using the Zelda tune. Nice work.

As you said, the issue itself doesn't seem to be reproducible, so I'll close this out assuming that it has been worked out for now. Since I don't have anything else to try/troubleshoot. :shrug:

closed 2023-11-18 by sntyj-dev

theorangerider, 2023-12-13:
I ran into the same issue and don't use totp, but do have the hourly chime with zelda. Thanks for the patch, @WesleyAC, I'll also give it a go.

theorangerider, 2023-12-13:
I wonder if the fact that it gets soft locked at midnight means it's conflicting with some other daily activity or default alarm.