joeycastillo/Sensor-Watch #331: Sensor-Watch Lite temperature offset

drtrigon, 2023-11-21:
Dear Sensor-Watch Makers

Thanks for this nice and clever product. I quite like it!

While using it as a thermometer I found that the temperature reported is about +1°C compared to other thermometers palced close to it.

Is there a easy way e.g. using preferences to calibrate respective set the temperature offset such that it shows the correct temperature?

Thanks and Greetings

bricewge, 2023-12-04:
I can confirm, the sensor watch lite report higher temperature than the actual temperature even after leaving it away of a wrist for some time.

If some settings have to be adjusted, it's probably these: 
https://github.com/joeycastillo/Sensor-Watch/blob/1b90a4a07a591c6f074eacb0f0dae9cabc7e0f90/watch-library/shared/driver/thermistor_driver.h#L34C27-L37

BrianBinFL, 2023-12-12:
> I can confirm, the sensor watch lite report higher temperature than the actual temperature even after leaving it away of a wrist for some time.

Of course there will be some delay for the temperature of the sensor inside the watch to rise and fall in response to changes in the environmental temperature and/or the presence or absence of a warm wrist. But I have generally found the watches to respond rather quickly to on/off wrist - likely due to the fact that the watches aren't very thick and the case back is metal.

I have two watches in a laboratory incubator for some rate experiments and keep the "thermistor face" active on them. The temperature shown on the watches is generally within about 0.4 degrees Celsius of the temperature the incubator is set to, and of course some of that will be variances in the actual temperature as the incubator works to keep it constant.

drtrigon, 2024-10-11:
Would it be possible to add configuration to the temperature watch face? E.g. an offset value that can be set for adjustment?

bricewge, 2024-10-11:
A few months ago I tried to reproduce the issue again, but I wasn't, this time the temperature reported by the watch left on a desk for a few hours was in sync with other thermometer in the room. I didn't update it since my first test in December of last year.

A week ago, in the [second project update for the Sensor Watch Pro](https://www.crowdsupply.com/oddly-specific-objects/sensor-watch-pro/updates/fully-funded-in-under-a-week-also-a-tour-of-second-movement-the-new-community-firmware-for-sensor-watch) I spotted this:

 > Things like cleaning up brittle ADC code for the temperature [...]

I didn't managed to find the commit(s) in question.