joeycastillo/Sensor-Watch #343: totp_face

Haifisch92, 2023-12-25:
hi i ve tried to deploy totp_face but output code on watch are different from app authenticator , so i have tried to change algorithm with sha256 without good result . in the end i have tried to flash firmware with the default TOTP secret and add TOTP secret on my authenticator app but code still not match ...

someone flashed this face with good result???

Haifisch92, 2023-12-25:
from emulator on https://sensor-watch-builder.fly.dev/ , output code match with google auth, i dont understand what i'm doing wrong

WesleyAC, 2023-12-25:
Have you set the time correctly on the watch, including the timezone offset? My guess would be that you didn't set the timezone.

Haifisch92, 2023-12-25:
yeah thanks a lot , the problem was timezone

closed 2024-01-09 by WesleyAC