joeycastillo/Sensor-Watch #348: TOTP no longer works as of 2 January 2024

matthewcmead, 2024-01-02:
I've had working a totp face since February 2023.  I haven't used the watch in a couple weeks for totp, and much to my surprise, it doesn't work today.  I've validated with another totp generator which has the same seeds that the values the watch is producing are different.  I used the watch for totp successfully three weeks ago.  The timezone is still set the same and the time is synchronized.

Any idea what might be going on?  Nothing's changed with the watch -- it's been sealed since I installed the firmware last February.

Thanks!

tahnok, 2024-01-02:
I think there was a bug in the timestamp calculation that was fixed, you
may want to see if upgrading your firmware fixes it

matthewcmead, 2024-01-03:
That did it.  Thanks!  Please feel free to close the issue.

joeycastillo, 2024-01-03:
I’d like to leave this issue open until I address it in the posted firmware images on sensorwatch.net

thanks so much to everyone for reporting and recognizing the issue!

EmilienCourt, 2024-01-07:
For completeness (and since I encountered this issue myself), this bug occurred on the 1st January of 2024, since it's a leap year and (as @tahnok pointed out) there was a bug in the timestamp calculation for those years.

The bug was raised [here](https://github.com/joeycastillo/Sensor-Watch/issues/261) and fixed [here](https://github.com/joeycastillo/Sensor-Watch/pull/262)

joeycastillo, 2024-01-09:
Thanks so much to everyone for identifying and addressing this bug, and sorry for being asleep at the wheel deploying the fix. As mentioned, this bug has been fixed in main; you will need to rebuild and reflash your board to get the fix. 

I've also updated all the [prebuilt firmware images on sensorwatch.net](https://www.sensorwatch.net/docs/firmware/prebuilt/); while none of these firmware images contained the TOTP function, some contained [timers](https://github.com/joeycastillo/Sensor-Watch/issues/353) and [countdowns](https://github.com/joeycastillo/Sensor-Watch/issues/352) that were affected. Reflashing with an updated firmware image should address the issue folks were seeing as we entered this leap year.

closed 2024-01-09 by joeycastillo