joeycastillo/Sensor-Watch #352: timer_face not working properly anymore

814d3, 2024-01-08:
I recognized, that timer_face shows weird output after the set time is over, additionally there is no beep/signal.
The lcd displays an output like this (example: time set to two minutes):

00:02 00
00:01 59
00:01 58
...
00:00 01
00:00 00 
(until here not noticeable, but **no** beep in the end)
(right after zero it displays this:)
00:00 25 
(for the next "steps": each step takes about 3 to X seconds)
00:00 24 
00:00 23
00:00 22
00:00 21
00:00 20
00:00 19
(then it seems stuck and shows:)
00:25 52
(after around 30 seconds it shows the following, but each "step" again takes 15 to X seconds:)
00:25 51
00:25 42
00:25 41
00:25 32
00:25 31
00:25 22
00:25 12
...

I used timer_face quiet regularly but somwhow it stopped working. Maybe it has something to do with turn of the year?

pfmaggi, 2024-01-08:
There was a bug in the timestamp calculation that was fixed, you may want to see if upgrading your firmware fixes it,

See #348

joeycastillo, 2024-01-09:
It looks like it was related to the leap year. If you were using a custom firmware, please rebuild and reflash with the latest code on main; if you were using one of the prebuilt alternative firmware images, you should be able to [download an updated version of it here](https://www.sensorwatch.net/docs/firmware/prebuilt/).

closed 2024-01-09 by joeycastillo