joeycastillo/Sensor-Watch #353: Tomato watch face not working anymore

h6y3, 2024-01-08:
I am not sure what happened, but it appears that the tomato watch face has stopped working correctly. When I start a pomodoro (either work or rest) the timer never ends, and it eventually starts displaying strange numbers as if it was rolling on itself. 

I pulled the battery hoping it would "fix" itself but it didn't so my next step is to just load up new firmware and hope it is fixed. 

Anybody have any ideas what could have caused this or if there was an update that would address this?

h6y3, 2024-01-09:
Here's a picture of the watch face which is in break mode and it's just doing all sorts of crazy things. :)

[image]

tahnok, 2024-01-09:
I think this is the timestamp/ leapyear bug again (see other recent issues)
that should be fixed by upgrading the firmware

joeycastillo, 2024-01-09:
It looks like it was related to the leap year. If you were using a custom firmware, please rebuild and reflash with the latest code on main; if you were using one of the prebuilt alternative firmware images, you should be able to [download an updated version of it here](https://www.sensorwatch.net/docs/firmware/prebuilt/).

closed 2024-01-09 by joeycastillo

h6y3, 2024-01-13:
Thank you! This makes sense and the timing corresponds with the breakage. 

I appreciate all of the support - it's an amazing platform!