joeycastillo/Sensor-Watch #355: Not entering LE mode under certain conditions

maxz, 2024-01-12:
There seems to be an undesired condition which prevents the watch from entering its low energy mode.

The only "unusual" thing I did yesterday was using the stopwatch. I measured something and stopped at exactly 10 minutes. After using it I reset it to 0 (pressing alarm once to stop it and then pressing light to reset it.) and changed back to the default face.
This morning when I woke up the watch was still in its normal mode. When I switched to the stopwatch face again, started it, stopped it and reset it again, the watch properly entered the LE mode after 1 hour.

This occurred with an image based on commit `63d6bc6aa0ddf4cc1ce1918ef7650852a25e581b`.

My used faces are 
```
simple_clock_face,
stopwatch_face,
totp_face,
preferences_face,
nanosec_face,
finetune_face,
set_time_face
```
Additionally I used the chime `SIGNAL_TUNE_ZELDA_SECRET`.

I will now flash my watch with an image based on commit `63d6bc6aa0ddf4cc1ce1918ef7650852a25e581b`, but wanted to document this before so that it will be clear that it's not due to the fix from #325.

After flashing the new firmware image I will use the stopwatch more often and try to provoke this issue to see under which specific conditions it happens.
Maybe someone else encountered this too and has observations.

814d3, 2024-01-20:
Any updates on this?

maxz, 2024-01-20:
Not really. So far it has not happened again but I also have not used the stopwatch since flashing a newer version.

maxz, 2024-01-31:
I experienced this again today, but I did not use any special faces.

The only thing special I did this time was deactivate the alarm some time yesterday but I reactivated it before I took off the watch for the night.

drtrigon, 2024-11-19:
I use another firmware with other faces but encounter the same issue. My firmware is similar to the default one, except:
- world_clock_face
+ beats_face
+ timer_face
+ wake_face
(I use the stock_stopwatch_face)

This means @maxz and I have in common the simple_clock_face, preferences_face and set_time_face. They all are included in the default firmware also.

@maxz: You mentioned "alarm" in your last posting, does that mean you also run a face like wake_face as well?

May be I should also mention that low energy used to work for me and my firmware but stopped a few days/weeks ago without having me to change habits or usage noticeably.

maxz, 2024-11-19:
No, only the mentioned faces, @drtrigon.

I probably meant the hourly chime.