joeycastillo/Sensor-Watch #357: The LE mode is canceld when you swipe your finger across the display

emloz, 2024-01-15:
Hi
Is the PTC Touch functionality active?
I see that the PA2 BTN-alarm is on pull-down including debounce. 
Is there a register code to clearly switch off PTC in the SAM L22.

Thank and best regards

emloz, 2024-01-15:
It could be static electricity from the display glass when you swipe over it.

WesleyAC, 2024-01-21:
The touch functionality is not something that could be supported on this hardware, and thus is not enabled in the firmware.

Is this something that you're seeing consistently, or only intermittently?

WesleyAC, 2024-01-22:
Very strange! Do you see this when you bump/shake it in other contexts, or just when you swipe your finger in that way?

tofublock, 2024-01-23:
Could this be a brownout reset due to the battery clip briefly losing contact or would this look differently?