joeycastillo/Sensor-Watch #359: Watch Freezes at Midnight in LE

Dabaraboo, 2024-01-17:
[image]
I've woken up to find the sensor watch frozen at midnight a couple times, its unresponsive to an alarm press to wake, and the second ticker is still running. I'm wondering if anyone else has experienced this issue?  
I've been using the unofficial firmware builder.

814d3, 2024-01-20:
How did you solve the issue? Was it because of an outdated firmware version?

Dabaraboo, 2024-01-22:
> How did you solve the issue? Was it because of an outdated firmware version?

Dabaraboo, 2024-01-22:
It may be solved with the updated firmware. closed the issue given the lack of relevant info I had. 
Accidentally closed it again just now because I have no idea what i'm doing, lol

closed 2024-01-22 by WesleyAC