joeycastillo/Sensor-Watch #363: Blank screen after update

pebblewidower, 2024-01-21:
I had the countdown leap year issue, so I decided to open the watch upgrade the firmware when I heard of the solution, something I haven't done for nearly a year. I also decided to put nanosec, finetune and time set on a secondary screen.

I fresh installed gh on a new computer and ran all the usual steps listed on the readme file plus COLOR=GREEN. However, no matter what I do, all I get is a blank screen.

P.S.: battery was at 2.48V today, so it's not a battery issue.

maxz, 2024-01-21:
It is most likely related to #339 and fixed in #362.

theAlexes, 2024-01-22:
yeah, that one's on me. sorry for the inconvenience.

tahnok, 2024-01-22:
Fix is merged

WesleyAC, 2024-01-22:
Eep! thanks for reporting this, and for the quick revert @tahnok.

pebblewidower, 2024-01-22:
I'm still having the same issue, even after a fresh new clone of the repo.

WesleyAC, 2024-01-23:
Uh oh! If you want to be safe, you could jump back to 07e45acba7c47748a382ef63ab35b04dd22bdc9b (or even 76b580a5be70df3e06dadb42fe328d66e1f83d31) and build that, and it should get you working.

If you want to help debug, running a `git bisect` or just stepping back each commit from `main` would be hugely helpful. I will do that if no one else gets to it by the time i get a chance to pull apart my watch, though.

maxz, 2024-01-23:
I actually cannot replicate this. I just built and flashed a firmware image based on commit `11de5936c51afe38cef28ecad4b038f35bd622e1`, with my TOTP patches on top and some other stuff like custom weekday names, but nothing that should affect anything related to this issue. It works just fine.

pebblewidower, 2024-01-23:
Tried `main`, `07e45ac`, `11de5936c51afe38cef28ecad4b038f35bd622e1`, with and without  `make COLOR=GREEN clean && make COLOR=GREEN`, to no avail.

However...

The red LED used to flash when double tapping reset with the battery attached, now it only does it when connected via USB. I decided to test the battery with the original Casio module and it doesn't turn on either. If I were to guess, it seems the problem is now related to battery and/or its attachments, due to the many tests I did.

I'll source a new battery and report.

pebblewidower, 2024-01-23:
Tried `main`, 07e45ac, `11de5936c51afe38cef28ecad4b038f35bd622e1`, with and without `make COLOR=GREEN clean && make COLOR=GREEN`, to no avail.

However...

The red LED used to flash when double tapping reset with the battery attached, now it only does it when connected via USB. I decided to test the battery with the original Casio module and it doesn't turn on either. If I were to guess, it seems the problem is now related to the battery and/or its attachments (although it does still click in place like always) due to the many tests I did.

I'll source a new battery and report.

pebblewidower, 2024-11-13:
I feel stupid writing this, but it seems the battery died between having the issue and its fix, even thought it showed a healthy 2.48V before dying a couple of days later.