joeycastillo/Sensor-Watch #377: timer_face sometimes no buzzing

814d3, 2024-03-06:
Hi there,

I use the timer_face regularly and recognized that sometimes it doesn't buzz after the time is over or if I start/stop and scroll through different presets.
What I can tell is that normal button pushes activate the buzzer normally. But when I use timer_face there is sometimes no beep. In addition there is no beep after the set timeout, when it switches back to simple_clock_face.

I'm not sure if I got it right, but there was one time I used timer_face but after the set time was over, there was no buzzing 5 times, but only 1 or 2 times. I can't  tell if the other 3-4 buzzes were just very quiet or nonexistent. But this reminded me of the general issue with the buzzer/hourly chime we had some weeks before.

matheusmoreira, 2024-03-08:
To me it sounds like there's a race condition in here somewhere. The watch face uses the following functions to activate the buzzer:

 - `watch_buzzer_play_note`
   - Synchronous
 - `watch_buzzer_play_sequence`
   - Asynchronous

They could get interrupted by ticks or user input. It's not immediately clear to me what's supposed to happen to the buzzer when that happens. It's possible that when we press buttons the buzzer state gets overwritten. What if that happens while it's in the middle of playing a sequence of notes? That operation probably gets cancelled.

The complexity of this project seems to be quickly approaching that of real time operating system.

814d3, 2024-03-09:
Well, I checked it again and I can somehow reproduce the behavior:
Just scroll through the presets fast and activate a timer and deactivate it afterwards (after ~1 second). (Maybe do it again) Then there is either no buzz/a short or aborted buzz or very quiet buzz when doing the above steps again.

But why it sometimes is not/very quiet buzzing if I just activate e.g. a 5 min. timer....idk

814d3, 2024-05-28:
I cleaned the buzzer contacts a while ago and somehow thought, that this has brought improvement. But I noticed it again: Set a 2 hours timer. After the time expired, there was only one really short buzz instead of 5 clear beeps...  :-(

814d3, 2024-06-29:
Thanks for your comment, so it seems to be no individual problem/cause. Hopefully someone could dive into it.
I'll try a W-59 to rule out that the cause lies with other hardware.

closed 2024-08-05 by 814d3