joeycastillo/Sensor-Watch #399: How to get serial output from USB?

RubenSandwich, 2024-05-07:
I just got my sensor watch a week ago and have already started developing for it. ([I just finished creating a watch face](https://github.com/RubenSandwich/Sensor-Watch/commit/fa35b8bb77f687fdd0eba09a6d8e3b49e72ab272).)

I am now interested in using the accelerator data from the `lis2dw` to create a simple pedometer using [existing open source algorithms](https://github.com/nerajbobra/embedded_pedometer). However because getting accurate reading from this sensor requires the actual watch board, and not just he simulator, I am struggling to debug certain parts of my code.

So my question is, how do I get serial output from the watch board when plugged into it with USB? (So I can view my `printf` statements.)

wryun, 2024-05-11:
It should come up on your computer as a standard serial port when you connect it via USB, and you can use the terminal program of your choice to communicate with it.

What OS are you using?

By the way, if you got your sensor watch a week ago I'm guessing you got the sensor watch lite, so it doesn't have the port to add an accelerometer. Not sure if you've already hacked your way around this :)

(PS might be good to open this sort of thing in 'Discussions' next time)

closed 2024-05-13 by RubenSandwich

RubenSandwich, 2024-05-13:
I was able to get serial output with minicom. I'm till new to embedded development, for forgiven my ignorance.

You're right that I have a sensor watch lite. When looking through the code I got confused between OSO-SWAT-B1 (the sensor watch lite board) and OSO-SWAB-B1-00 (the upcoming sensor board) which has the LIS2DW12TR accelerometer on it.

It does look like it _might_ be possible to wire the sensor watch lite to the LIS2DW12TR sensor board with the pins:
* 59 - SDA P (as used in the original sensor watch, but unused on sensor watch lite)
* 60 - SCL (as used in the original sensor watch, but unused on sensor watch lite)
* 61 - A4 pad
* 64 - CS (as used in the original sensor watch, but unused on sensor watch lite)
* 3v - through debugging pins
* GND - through debugging pins

However, I think micro soldering to microcontrollers unused pins is currently beyond my soldering ability.