joeycastillo/Sensor-Watch #407: Yellow LED constantly on for red board

rieck, 2024-06-16:
The yellow LED on my red board stays on continuously, which typically signals an incorrect firmware version (color). However, I have compiled the firmware correctly using `make COLOR=red`, and the watch is actuially functioning normally. The only issue is that the yellow LED remains lit. It’s not due to a short circuit from the case, as the LED remains on even when connected to USB. Does anyone have any ideas about what might be causing this?

joeycastillo, 2024-06-16:
Alas the issue is with uppercase vs lowercase: you have to `make COLOR=RED` for it to work. 

Is there some make-fu we can use to make [these lines](https://github.com/joeycastillo/Sensor-Watch/blob/main/make.mk#L220) case insensitive? Or uppercase whatever the input is?

rieck, 2024-06-16:
Oh, thanks for the swift response!! 

`COLOR := $(shell echo $(COLOR) | awk '{print toupper($$0)}')`

or

`COLOR := $(shell echo $(COLOR) | tr '[:lower:]' '[:upper:]')`

would be two options. They require the `shell` subcommand, which is not present in BSD make to my knowledge, and likely they both do not work on Windows.

wryun, 2024-06-16:
I think the best thing to do is to have it abort if the COLOR line has an unrecognised colour.

chris-v8, 2024-06-16:
or modify the makefile to skip the COLOR argument, and always run it with your watch colour

orhun, 2024-06-27:
I'm having the same issue. Is there way to turn this off for the pre-built firmware? Do I need to build the firmware with the flag?