joeycastillo/Sensor-Watch #409: Flashing firmware instructions for the command line

orhun, 2024-06-26:
Hello!

Is there any guide for flashing firmware from the terminal? I checked [the documentation](https://www.sensorwatch.net/docs/firmware/flashing/) but unfortunately it is not suitable for my case.

Thank you!

matheusmoreira, 2024-06-26:
Hello! Are you running Linux?

Linux should automatically detect and expose the watch to the system as a mass storage device. Then you can mount it and copy the firmware to the root directory of the device. That will flash the firmware.

I wrote a simple script to do it automatically. It's on my laptop and I'm at work right now. I'll share it here when I get back home.

orhun, 2024-06-27:
Hey, that worked thanks! I simply `mount`ed it and copied the firmware :)

closed 2024-06-27 by orhun