joeycastillo/Sensor-Watch #415: Battery drain

simon-src, 2024-07-07:
I have the green Sensor-Watch board and have been experiencing extreme battery drain ever since i first installed it in January 2023. I have done the following steps to troubleshoot the issue:

- Tried different battery brands: Panasonic and Duracell, all brand new and measured before install
- Removed thermistor
- Tried different screens
- Tried different housings
- With piezo contact soldered
- Without piezo contact soldered
- With another piezo metal contact soldered
- With another battery metal spring thing
- With only minimal faces (watch, volt, countdown, settings) and always latest clean main pulled

Battery is always down to 1.5V within 60 to 70 days. If i leave it in longer it will end up 0V

What other things can I try? I can't see anything obviously wrong on the board.

814d3, 2024-07-08:
Maybe you could measure the current that the controller/board is drawing?
The watch should run until ~2,2 V. What do you mean with 1,5 V?

shmoun, 2024-07-08:
Thanks for your suggestion! I will try to measure the current drawn and come back with a number.

> What do you mean with 1,5 V?

The ~1.5V reading i got from the voltage face, the lowest reading I got before the display went blank. It was already hovering around 2V a bit below and then a bit above. But when trying the light while in the voltage face it dived to 1.5V display blanking and never got back. 

That was yesterday evening (17 hours ago).  Now I took out battery and it seems to have continued draining, my multi meter measures 0.10V

shmoun, 2024-07-08:
My cheap multimeter reading for 30 seconds after boot is as follows (multimeter set to 2000uA):
Boot -> 380
Wake operation -> varying rhytmically between 010, 012, 019

I didn't have the patience and steady hand to get to LE mode (yet)

814d3, 2024-07-08:
I remember runtime measurements done by Joey while testing and optimizing the green boards. Maybe you can find those and compare with your values.

shmoun, 2024-07-10:
Thank you @814d3 

I found the Sensor-Watch discord and someone reporting what I'm seeing with the same board:
https://discord.com/channels/969627287429459988/1029086310192918538/1107111421940076555

>I'll dig up the message chain... in particular I (and I think another person) were observing fresh coin cells deplete in ~30 min to 1 hour (on the voltage face I saw the battery start well above 3V and quickly tank to, like, 2.7V, confirmed with multimeter)

I see that behavior, the battery starts at 3V and goes down in the first hour(s) to the 2.7V.

shmoun, 2024-07-16:
My battery drain issue is fixed \o/ 

Using a tiny USB magnifying camera I noticed a lot of chunky pieces of what i think is flux residue spreaded over the board. The biggest chunk was covering a large part of led. It was such a big piece that on previous inspections i thought it was some kind of hard epoxy cover to protect the led, but when carefully prying it with plastic tweezers it moved and was soft and rubbery. Very sticky too. I removed all the big blobs i could find from the board. 

Battery drain has stopped, I can actually watch the battery recovering, voltage increasing since yesterday when i did the clean-up.

I included a picture of the LED blob after i already loosened it and moved it to the side of the LED.
[image]

closed 2024-07-16 by shmoun

nrob81, 2024-08-04:
Hello, I have the same problem now. I have my clock one week now and my first battery drained in ~6 days. Yesterday I replaced my battery. It has 3.30V, today the clocks voltage meter displayed 3.13V. I used custom FW with simple clock, sun times, moon phases, countdown, tomato, alarm, thermistor readout, voltage. today replaced it to prebuild FW Focus. Hope it helps. I did not find anything on my board with microscope.