joeycastillo/Sensor-Watch #418: Watch resets every 2 days

sajingeo, 2024-07-11:
Here is my config and my watch resets every 2 days. 

```
const watch_face_t watch_faces[] = {
    simple_clock_face,
    stopwatch_face,
    alarm_face,
    world_clock2_face,
    sunrise_sunset_face,
    counter_face,
    timer_face,
    totp_face,
    preferences_face,
    set_time_face,
    thermistor_readout_face,
    voltage_face
};
```
battery voltage is at 3.2 v (tried with the new battery - same issue).  I don't think it is a hardware issue, I have tried to apply some force on the watch to see if the power drops out or tried to recreate the resets with button press or motion. It only happens at night, since I always see the time reset first thing in the morning. So the reset is happening when I am not actively using the watch.

shmoun, 2024-07-13:
In the original F91W watches I've opened I have found a tiny piece of good quality double sided tape/sticky material keeping the battery in place. I put a piece of 3M double sided adhesive ~2mm width when installing fresh batteries. 

Another thing to pay attention to is the little clips of the metal cover, are they really all settled over the plastic base? Since the fit of the board is tight, it needs some extra attention to fit the clips properly.

On the software side you can start with excluding watch faces one by one. Or start with the most minimal set of faces.

sajingeo, 2024-07-16:
Its fixed now... not exactly sure what fixed it here are all the steps.

1. Removed the tape under the battery and put a new one in
2. cleaned all the contacts with alcohol
3. removed the totp_face

Watch has been running for 3 days wihout any issues.

closed 2024-07-16 by sajingeo