joeycastillo/Sensor-Watch #485: Compatibility

supersarma, 2024-10-23:
Hi there,

I bought a  A1000M-1BEF with the same module as f-91.
I would like to know if this projet is compatible with this case ?

Thank you for your help.

CarpeNoctem, 2024-10-24:
Are you sure it contains the same module as the F91? The case seems to say Illuminator on it, which makes me think it probably uses a different module and would be incompatible. There are many Casio models that share the same shape and button layout but use a different (incompatible) module from the F91.

The list of known compatible and incompatible models (checked so far) is here: https://www.sensorwatch.net/docs/#need-to-buy-a-watch

You could buy a Sensor Watch (Lite) and give it a try; and if it didn't work with that model, you could buy one of the other supported models to put your Sensor Watch into. Before even buying a Sensor Watch, you could also open the your A1000M up and see whether it even looks close enough to what you see in the installation guide.

We'd love to hear how you go with any of the above! The A1000M is a great looking watch. :)

supersarma, 2024-10-26:
Oh. Sorry for my ignorance. I indeed though that the module used in the watch is basically based on the shape. As the A1000 and F-91 share the same shape and are pretty same in the menus I though modules are same. 
After a check, modules are completely different :)

I ordered the board yesterday and as I live in France it will probably take some time.
As soon as I got it I will check  if it fits in the A1000. It would be great. I appreciate the f-91 but the A1000 looks reaaally nice.

CarpeNoctem, 2024-10-27:
If the watches are exactly the same dimensions inside, you _might_ be able to get the screen part from and F-91W and put that + the sensor Watch into the A1000.

814d3, 2024-10-27:
> The list of known compatible and incompatible models (checked so far) is here: https://www.sensorwatch.net/docs/#need-to-buy-a-watch

Well, the sensor board is also compatible to the W-59, which is not listed there.

kevin1024, 2025-07-28:
Did it work?  Was the a1000 case compatible?