joeycastillo/Sensor-Watch #491: Sunrise / sunset is incorrect for westerly timezones

edwardspeyer, 2024-11-06:
Sunrise and sunset are calculated by calling `sun_rise_set` for the UTC date instead of the local date.  This means that the next pair of rise/set times may be for tomorrow instead of the upcoming sunset.

Example: The watch is in San Francisco and the local time is 6pm on October 31st.  The sun will set in 12 minutes so the `sunrise_sunset_face` _should_ show the next pair of events as being (1) the time of today's sunset (6:12pm Oct 31st) followed by (2) tomorrows sunrise (7:37am Nov 1st).  However, the date is calculated from the UTC time which is 7 hours ahead in November 1st.   The face incorrectly shows the next pair of events as (1) an upcoming sunrise (7:37am Nov 1st) followed by (2) the next sunset (6:12pm Nov 1st).

edwardspeyer, 2024-11-06:
I think the fix for this is as simple as follows?:

```patch
--- a/movement/watch_faces/complication/sunrise_sunset_face.c
+++ b/movement/watch_faces/complication/sunrise_sunset_face.c
@@ -64,7 +64,7 @@ static void _sunrise_sunset_face_update(movement_settings_t *settings, sunrise_s
     watch_date_time date_time = watch_rtc_get_date_time(); // the current local date / time
     watch_date_time utc_now = watch_utility_date_time_convert_zone(date_time, movement_timezone_offsets[settings->bit.time_zone] * 60, 0); // the current date / time in UTC
     watch_date_time scratch_time; // scratchpad, contains different values at different times
-    scratch_time.reg = utc_now.reg;
+    scratch_time.reg = date_time.reg;
 
     // Weird quirky unsigned things were happening when I tried to cast these directly to doubles below.
     // it looks redundant, but extracting them to local int16's seemed to fix it.
```

vagrantc, 2024-12-12:
Had the same issue, and your patch worked for me!

edwardspeyer, 2024-12-12:
@joeycastillo sorry to not raise this as a pull request; would you consider the above is patch?

vagrantc, 2024-12-13:
Might be easier to just submit a pull request than waiting for a response. :)