joeycastillo/Sensor-Watch #501: battery drained in several days on sensorwatch light board

vagrantc, 2024-12-12:
I finally installed my sensorwatch light, but the battery only lasts maybe 3-4 days.

At first, I just had the default firmware, but have since tried some of the prebuilt images and also a custom image.

I just swapped in a fresh battery last night, which was reading over 3.2v when i installed it, but as of this morning, it is down to 3v.

The seconds display does change behavior which *might* indicate it is going to low-power/sleep mode, although there is constant activity on it... e.g. not displaying numbers, but displaying some segment changes every second.
so i am not sure how low power that actually is.

Any thoughts or suggestions?

joeycastillo, 2024-12-12:
No worries! Your battery is doing fine. It’s normal for the battery voltage to decline from 3.2 volts to 3 volts over the first few days. It’ll then remain at 2.9-3.0 volts for six to nine months — that’s the first half of the battery’s life — before beginning a decline to 2.9, 2.8 and then 2.7 volts; that’s the normal discharge curve for a 3 volt coin cell. Below 2.65 volts you’re living on borrowed time, but that shouldn’t happen for many, many months :)

The change in the seconds display is also normal: it’s the watch going into low energy mode, which by default happens after an hour without pressing any buttons. You can wake it up by pressing the bottom right button, and configure the low energy timeout in the “LE” settings screen.

vagrantc, 2024-12-13:
I did have the first battery drained down all the way to 0V, though I did not measure the voltage on that one at the start, as it was from a newly purchased watch... so I guess that is why I am a bit nervous.

That said, we'll see how long this new one lasts... if it doesn't plummet wildly outside of the curve you describe in the next couple days I guess I will know.

:)

Thanks!