joeycastillo/Sensor-Watch #506: Screen displays odd data after red board swap

whyleyc, 2025-02-13:
I recently bought a red board and followed [the tutorial](https://www.youtube.com/watch?v=Zr0pKeC2VFU) to swap out the buzzer connector and board in place of a standard F-91W version, but ran into some issues after re-assembling the watch. Specifically the display does not show any meaningful data:

[image]

To rule out any firmware issues I haven't installed any custom software on there, just the base ["Movement Standard" for red boards](https://www.sensorwatch.net/docs/firmware/download/standard-red.uf2) (I have also tried installing custom firmware with the same resulta).

The LED works fine:

[image]

And it is able to scroll through "modes", but doesn't show anything of use when doing that. It also appears to be possible to put it into the mode to set the time (or some version of it!):

https://github.com/user-attachments/assets/3a1ce4bc-fc0a-499c-8b4f-d5a69af6b9e2

I swapped the standard F-91W board back in thinking it might be an issue with the display, but this works fine - I can see all the usual functions on the display exactly as before the swap.

Here is the board in various stages of construction in case anyone is able to spot anything odd in the way I'm putting it back together (or any defects in the board itself):

**Side 1:**
[image]

**Side 2 (with battery connector):**
[image]

**With case:**
[image]

**With battery:**
[image]

**Assembled Back:**
[image]

**Assembled Front:**
[image]

Does anyone know what could be causing this issue or how to resolve? Could it be a defective board?

whyleyc, 2025-02-13:
> You have used a counterfeit watch. They are not compatible with the sensor watch board.

I'm not so sure - the watch came with Casio tags and the standard Casio F-91W branded manual - what makes you think the watch is a counterfeit?

Yometalc, 2025-02-13:
The pink zebra strip and the issue you have with the LCD.  The counterfeits are very good now but there are tell tale signs when you know what to look for.  Where did you purchase it? I see you are in the UK, Argos sells genuine ones for £11…

whyleyc, 2025-02-13:
Replacing the watch worked - I grabbed a new one from Argos, popped the board into it and everything sprang into life, so it looks like the first watch was indeed a fake.

Thanks for your assistance @Yometalc !

closed 2025-02-13 by whyleyc